markniu/PandaZHU #1: ADC question

luc-github, 2021-07-27:
Hi nice board, 
Any plan to use external ADC instead of ESP32 ones to avoid the 30~250 degres limitation ?

markniu, 2021-07-28:
Thanks! and your ESP3D project is great! and is really useful.

I think it's no need to use external ADC,I find most of the printer's temperature is not accurately at above 260 degree because of the 100K thermistor. so if we want to print at higher temperture, the best way is PT100 with MAX31865.

luc-github, 2021-07-28:
Thank you ^_^
one reason also is the failure detection (no thermistor error is not detected) , and room temperature detection is always incorrect which is confusing for some end users.
I agree they are edge case and external adc are nice to have, but I got some safety concern feedback so better ask. 😸 I heard new S3 adc are better,TBC

I plan to order one of your board but will be in october after my moving .

closed 2021-07-28 by luc-github

markniu, 2021-07-29:
Hi Luc, I would like to send one board to you for your develop, you can send me(niujl123@sina.com) your shipping address. 
> I plan to order one of your board but will be in october after my moving .

luc-github, 2021-07-29:
Hi Mark, I just sent you an email ^_^