markniu/PandaZHU #4: Regarding Pin Mapping

itskskalsi, 2022-01-28:
Can you please tell me how the pins of ESP32 are mapped to driver outputs, heaters and fans i.e in schematic I can see them connected through STC15W408AS controller which is communicating with esp32 using RX/TX lines (probably UART). And in code pins of driver outputs like Step,DIR,Enable are labelled as 100,104,108,etc.
So can you please explain me how they are connected and how will I get output on the drivers using esp32.

[image]

itskskalsi, 2022-01-28:
Also I want to know where in code can i find the communication part of ESP32 with STC15W408AS.

pegasus-automation, 2022-03-28:
Did you find the firmware for STC15W408AS?

itskskalsi, 2022-03-28:
No i didn't

itskskalsi, 2022-05-03:
> Did you find the firmware for STC15W408AS?

Where can I find the the firmware for STC15W408AS