markniu/PandaZHU #8: Code for GPIO Expander IC

pegasus-automation, 2022-03-28:
Hello I checked your board. It is really great. I wanted the code that we need to upload on IO Expander STC15W408AS IC. As the code that needs to be uploaded on ESP32 is available, but I could not the code for IO Expander IC STC15W408AS

Dschorim, 2022-08-21:
I was wandering the same thing, I want to modify the design to be compatible with the new ez-drivers, but in order to do this I'd need the code for the STC15W408AS

markniu, 2025-09-01:
Sorry, I just uploaded the code for this GPIO expander. This board is out of stock and hasn't been maintained in a long time.

rig-rigos, 2025-09-01:
> Sorry, I just uploaded the code for this GPIO expander. This board is out of stock and hasn't been maintained in a long time.

It's a pity. GPIO code for what option? Pandam4 or Pandazhu. I see they are different.
Thank you.

rig-rigos, 2025-09-06:
Ok, thanks for the answer. I understand that the chip is the same. The question is about something else. In the circuit diagram, the ZHU chip pins and the M pins control different signals.
Do I understand correctly that the GPIO expander code works correctly in both versions?

[image]